FreeRTOS/FreeRTOS-Kernel #1213: [Feature Request] Add titles for each column of "vTaskListTasks"

wwhheerree, 2024-12-23:
**Is your feature request related to a problem? Please describe.**
When I use "vTaskListTasks" and print out current task report, there is no title for each column. 
For new beginners, they have to refer to FreeRTOS document for detailed decriptions of each column.
[image]

**Describe the solution you'd like**
I think adding a title for each column will be better for human readability.
[image]

I can help to create a PR for "task.c" if you think this enhancement is helpful.

aggarg, 2024-12-23:
While this is a good suggestion, it may break existing applications using this API as it increases the required buffer size. Additionally, you can print this line in application if you want. How about adding this information in the function documentation - https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/task.h#L2186?

wcai6, 2024-12-24:
OK. Can we add the below description to task.h comment? 
If yes, I will submit a PR. 

TaskName   State   Priority   HighWaterMark    TaskID
Task A            X          2               67                      2
Task B            B          1                188                   3
IDLE               R          0               67                      5

aggarg, 2024-12-24:
Yes, that sounds good. Please raise a PR.

wwhheerree, 2024-12-24:
PR has been submitted. 
https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/1215

aggarg, 2024-12-24:
I have merged the PR #1215. Thank you for your contribution!

closed 2024-12-24 by aggarg