FreeRTOS/FreeRTOS-Kernel #1220: [BUG] RP2350 port is in the wrong subdirectory vs the kernel import CMake script

mcknly, 2025-01-09:
**Describe the bug**
Support for RP2350 was merged into the [FreeRTOS-Kernel-Community-Supported-Ports](https://github.com/FreeRTOS/FreeRTOS-Kernel-Community-Supported-Ports/tree/3c5bfb8f2e557735b5200176b4a8b25a40c68d1b) submodule, which originated from [this commit](https://github.com/raspberrypi/FreeRTOS-Kernel/commit/4f7299d6ea746b27a9dd19e87af568e34bd65b15) on RPi's FreeRTOS fork (but was actually merged from @kilograham's fork of community ports [here](https://github.com/kilograham/FreeRTOS-Kernel-Community-Supported-Ports/tree/rp2350)). However, adding the port into the nested submodule directory breaks the kernel import CMake script which assumes it is at the same location as the RP2040 port, `portable/ThirdParty/GCC/`, which would make sense. Symlinking the RP2350 port directory into the same place as the RP2040 port is a workaround for this issue. Ideally the RP2040 and RP2350 ports should be in the same spot.

**Target**
- Development board: Raspberry Pi Pico2
- Instruction Set Architecture: ARM Cortex-M33
- IDE and version: VScode 1.96.2
- Toolchain and version: gcc-arm-none-eabi 12.2.1

**Host**
- Host OS: Linux
- Version: Debian 12

**To Reproduce**
- Pull latest `main` of this repo
- Update submodules `git submodule update --init`
- In a project, use `set(freertos_port_path "portable/ThirdParty/Community-Supported-Ports/GCC/RP2350_ARM_NTZ/FreeRTOS_Kernel_import.cmake")` for RP2350 port
- Configure project using CMake

**Expected behavior**
Successful configuration (RP2350 port is found)

**Screenshots**
Error message from CMake: 
```
[cmake] Using FREERTOS_KERNEL_PATH from environment ('/projects/FreeRTOS-Kernel')
[cmake] CMake Error at /projects/FreeRTOS-Kernel/portable/ThirdParty/Community-Supported-Ports/GCC/RP2350_ARM_NTZ/FreeRTOS_Kernel_import.cmake:66 (message):
[cmake]   Directory '/projects/FreeRTOS-Kernel' does not contain a
[cmake]   'rp2350-arm-s' port here: portable/ThirdParty/GCC/RP2350_ARM_NTZ
```

**Additional context**
The issue was originally produced in my [Breadboard-OS](https://github.com/mcknly/breadboard-os) project, and I have also been tracking the issue [here](https://github.com/mcknly/breadboard-os/issues/32). Building the project on [this branch](https://github.com/mcknly/breadboard-os/tree/32-switch-back-to-mainline-branches) is another way to reproduce this issue.

mcknly, 2025-01-10:
Some more information on this - I noticed that the RP2040 kernel import script had already been updated to dynamically look for RP2040/RP2350_ARM_NTZ/RP2350_RISCV both inside and outside the submodule. I assume the intention was to also include this more dynamic script within the community supported ports submodule as well. I opened a PR - FreeRTOS/FreeRTOS-Kernel-Community-Supported-Ports#18 - probably should get @kilograham's take on it.

kstribrnAmzn, 2025-01-13:
Thanks for taking the time to report this! I see you've raised a [corresponding PR](https://github.com/FreeRTOS/FreeRTOS-Kernel-Community-Supported-Ports/pull/18). I'll take a look at it and let you know if I have any feedback.

aggarg, 2025-01-14:
Thank you for your contribution!

closed 2025-01-14 by aggarg