FreeRTOS/FreeRTOS-Kernel #1242: struct sio not found

carlosdelfino, 2025-02-07:
When I try to compile FreeRTOS without using the suggestions in the README.md file of this portable, I get an error message saying that `sio` is not declared. I have tried several alternatives and none of them have been successful.

I am not using the guidelines in the README.md file because I have not had any success with it either.

kar-rahul-aws, 2025-02-07:
Hi @carlosdelfino 
I am able to build the CORTEX_M0+_RP2040 [demo](https://github.com/FreeRTOS/FreeRTOS-Community-Supported-Demos/tree/272ca5bfbe0ea8a5e24a19a4a0e2bd4bf00b4066/CORTEX_M0%2B_RP2040) with the GCC/RP2040 port.

The RP2040 HAL definitions are often enabled through `PICO_SDK_PATH` in the build system. Ensure your environment is correctly set up. If you are using a separate CMakeFile , make sure your **include directories** point to the RP2040 SDK headers.

Can you give this demo a try and see if that builds for you ?

carlosdelfino, 2025-02-07:
hi @kar-rahul-aws 

If you can help me, I have put the project in the repository, and my deadline is this Sunday.

So far I have managed to compile it by adding the `sio.h` library to `port.c`, but the firmware freezes when initializing the scheduler.

https://github.com/ArvoreDosSaberes/magnectostricao

aggarg, 2025-02-10:
As @kar-rahul-aws mentioned, you should start with [ this demo](https://github.com/FreeRTOS/FreeRTOS-Community-Supported-Demos/tree/272ca5bfbe0ea8a5e24a19a4a0e2bd4bf00b4066/CORTEX_M0%2B_RP2040) and then make the changes as per your needs.

I am closing this issue. Feel free to reopen if you are not able to build the above demo.

closed 2025-02-10 by aggarg