FreeRTOS/FreeRTOS-Kernel #1244: [Feature Request] Please add Release version tags with clear alignment to FreeRTOS Kernel versions

jondurrant, 2025-02-11:
**Is your feature request related to a problem? Please describe.**
Currently it is only possible to link to the head of the repo or an unlabelled change. This makes it dificult to document the version of the source that I have used. It also makes it difficult to work out which version of the comments online or documentation on FreeRTOS Kernel we should take read.

**Describe the solution you'd like**
Tagged release produced on GitHub to allow a build to checkout a specific version. Particularly when linking to this library as a submodule.

**Describe alternatives you've considered**
Ignoring this and using the better release strategy provided by FreeRTOS themselves.

**How many devices will this feature impact?**
Every project will benefit with a non breaking change.

**What are your project timelines?**
Currently, for commercial projects I am forced to recomend using FreeRTOS Kernel as the release strategy here could cause us to trip.  Ongoing.

**Additional context**
Add any other context or screenshots about the feature request here.

If you have the same (or similar) feature request, please upvote this issue with thumbs up 👍
and use the comments section to provide answers to the questions above.

jondurrant, 2025-02-11:
Sorry trying to post this on Raspberry PI. Appolagies.

closed 2025-02-11 by jondurrant

TheZitroX, 2025-02-26:
Hey — Did you tried the `Transfer Issue` to other Repository? Maybe this would help