FreeRTOS/FreeRTOS-Kernel #1263: [BUG]taskEVENT_LIST_ITEM_VALUE_IN_USE

Linux-yzy, 2025-04-09:
**Describe the bug**
The xTaskPriorityInherit function has the following judgment:
 if( ( listGET_LIST_ITEM_VALUE( &( pxMutexHolderTCB->xEventListItem ) ) & taskEVENT_LIST_ITEM_VALUE_IN_USE ) == 0UL )
 {

 }
#define taskEVENT_LIST_ITEM_VALUE_IN_USE    0x80000000UL
However, in the queue function, even if the tasks are blocked, the value of xItemValue in xEventListItem is always less than 32, which causes the above judgment to always be true.

**Screenshots**

[image]

[image]

**Additional context**
Add any other context about the problem here.
e.g. code snippet to reproduce the issue.
e.g. stack trace, memory dump, debugger log, and many etc.

aggarg, 2025-04-09:
Are you facing an issue or is your conclusion based on code reading?

Linux-yzy, 2025-04-09:
> Are you facing an issue or is your conclusion based on code reading?

I based it on the version FreeRTOSv202210.01-LTS

aggarg, 2025-04-09:
> I based it on the version FreeRTOSv202210.01-LTS

Did you face an error while running a program? If yes, can you share the application code and tell us what is the expected and actual behavior?

Linux-yzy, 2025-04-10:
> > I based it on the version FreeRTOSv202210.01-LTS
> 
> Did you face an error while running a program? If yes, can you share the application code and tell us what is the expected and actual behavior?
I didn't run the program, I found the problem when I was studying your kernel program

Linux-yzy, 2025-04-10:
I didn't run the program, I found the problem when I was studying your kernel program

kar-rahul-aws, 2025-04-11:
The `else` check in `xTaskPriorityInherit()` exists to avoid corrupting FreeRTOS event lists. When a task holds a mutex and gets its priority boosted, FreeRTOS updates its `xEventListItem->xItemValue` only if that item isn’t already linked to an event list (like when waiting on an EventGroup or Queue). If the item is linked, changing its value would break list ordering and corrupt the kernel’s internal structures. In that case, only the task’s `uxPriority` is updated — which is enough for correct scheduling, since ready lists use a separate `xStateListItem`.

See the below code as an example :- 

```

    /* Priorities */
    #define TASK_A_PRIO (3)
    #define TASK_B_PRIO (2)
    #define TASK_C_PRIO (1)
    
    /* Event Group bit */
    #define BIT_0 (1 << 0)
    #define BIT_1 (1 << 1)

    StaticTask_t taskAControl, taskBControl, taskCControl;
    StackType_t taskAStack[configMINIMAL_STACK_SIZE];
    StackType_t taskBStack[configMINIMAL_STACK_SIZE];
    StackType_t taskCStack[configMINIMAL_STACK_SIZE];

    StaticEventGroup_t eventGroupBuffer;
    EventGroupHandle_t xEventGroup;

    StaticSemaphore_t mutexBuffer;
    SemaphoreHandle_t xMutex;

    void TaskA( void * params )
    {
        for( ;; )
        {
            sci_print( ( "TaskA: Waiting for event bit\n" ) );
            xEventGroupWaitBits( xEventGroup, BIT_0, pdTRUE, pdFALSE, portMAX_DELAY );
            sci_print( ( "TaskA: Event received, trying to take mutex\n" ) );

            xSemaphoreTake( xMutex, portMAX_DELAY );

            sci_print( ( "TaskA: Acquired mutex!\n" ) );
            xSemaphoreGive( xMutex );
        }
    }

    void TaskB( void * params )
    {
        sci_print( ( "TaskB: Taking mutex\n" ) );
        xSemaphoreTake( xMutex, portMAX_DELAY );

        for( ;; )
        {
            sci_print( ( "TaskB: Waiting for event bit\n" ) );
            /* Block on BIT_1, so xEventListItem gets used */
            xEventGroupWaitBits( xEventGroup, BIT_1, pdTRUE, pdFALSE, portMAX_DELAY );

            sci_print( ( "TaskB: Woke up from event wait\n" ) );
            vTaskDelay( pdMS_TO_TICKS( 1000 ) );
        }
    }

    void TaskC( void * params )
    {
        for( ;; )
        {
            vTaskDelay( pdMS_TO_TICKS( 500 ) );
            sci_print( ( "TaskC: Setting event bit\n" ) );
            xEventGroupSetBits( xEventGroup, BIT_0 | BIT_1 );
        }
    }

    void app_main( void )
    {
        xEventGroup = xEventGroupCreateStatic( &eventGroupBuffer );
        xMutex = xSemaphoreCreateMutexStatic( &mutexBuffer );

        xTaskCreateStatic( TaskA, "TaskA", configMINIMAL_STACK_SIZE, NULL, TASK_A_PRIO, taskAStack, &taskAControl );
        xTaskCreateStatic( TaskB, "TaskB", configMINIMAL_STACK_SIZE, NULL, TASK_B_PRIO, taskBStack, &taskBControl );
        xTaskCreateStatic( TaskC, "TaskC", configMINIMAL_STACK_SIZE, NULL, TASK_C_PRIO, taskCStack, &taskCControl );

        vTaskStartScheduler();

        for( ;; );
    }

```

If the **else check wasn’t there**, this would corrupt the event list TaskB is waiting on in the following way -

1. TaskB owns the mutex → its priority needs to be boosted when TaskA tries to xSemaphoreTake().
2. But TaskB is _also_ blocked on `xEventGroupWaitBits()` (BIT_1) → so its `xEventListItem` is linked inside the event group's list.
3. Now, `xTaskPriorityInherit()` would overwrite `xEventListItem.xItemValue` with `configMAX_PRIORITIES - TaskA's priority`.
4. Problem: The event group's wait list uses `xItemValue` to maintain ordering → so this overwrite breaks list integrity.
5. Result: When the event group tries to unblock tasks later → corrupted list → unpredictable behavior → most likely a crash or assert in list.c (`configASSERT( pxList->xListEnd.pxNext == ( ListItem_t * ) &( pxList->xListEnd ) );`)

Hence, this **else check** protects kernel data structures when a mutex holder is also waiting on something else (like an EventGroup or Queue).

aggarg, 2025-04-12:
@Linux-yzy - Closing this issue as @kar-rahul-aws has provided a detailed explanation. Feel free to reopen if you still face any issue.

closed 2025-04-12 by aggarg