FreeRTOS/FreeRTOS-Kernel #1265: [BUG] Port MPLAB PIC32MZ-EF, Timer1 is 16bits

KammutierSpule, 2025-04-17:
**Describe the bug**
On the Port MPLAB PIC32MZ-EF, the Timer1 is 16bit

**Target**
- MPLAB PIC32MZ-EF

**To Reproduce**
- using configPERIPHERAL_CLOCK_HZ = 100 000 000 and configTICK_RATE_HZ = 100, the ulCompareMatch on vApplicationSetupTickTimerInterrupt will overflow. While using the current portTIMER_PRESCALE and portPRESCALE_BITS settings on the source code.

**Expected behavior**
- Expect that the vApplicationSetupTickTimerInterrupt asserts the ulCompareMatch to fit on a 16bit register.

aggarg, 2025-04-21:
Thank you for raising this. If I understand correctly, you want to add the following assert:
```c
    /* PR1 is 16-bit. Ensure that the configPERIPHERAL_CLOCK_HZ and
     * configTICK_RATE_HZ are defined such that ulCompareMatch value would fit
     * in 16-bits. */
    configASSERT( ( ulCompareMatch & 0xFFFF0000 ) == 0 );
```

If yes, we can add that. Would you like to raise a PR for that?

aggarg, 2025-04-24:
> the timer configuration is performed before the scheduler start (makes sense)

The timer configuration happens in the function vApplicationSetupTickTimerInterrupt which is called [when starting the scheduler](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/MPLAB/PIC32MZ/port.c#L286).

> the taskENTER_CRITICAL asserts because scheduler is not working.

On this port, the `taskENTER_CRITICAL` ends up calling [vTaskEnterCritical](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/tasks.c#L6989) and I am not sure why would it assert. Can you share the assert that you are hitting?

KammutierSpule, 2025-04-24:
I did some more investigation, please check the code below for `<--- FAILS!`

```
void vAssertCalled( const char * pcFile, unsigned long ulLine )
{
   taskENTER_CRITICAL();   <--- FAILS HERE!
   {
      // your code
   }
   taskEXIT_CRITICAL();
}
```

```
    void vTaskEnterCritical( void )
    {
        traceENTER_vTaskEnterCritical();

        portDISABLE_INTERRUPTS();

        if( xSchedulerRunning != pdFALSE )
        {
            ( pxCurrentTCB->uxCriticalNesting )++;

            /* This is not the interrupt safe version of the enter critical
             * function so  assert() if it is being called from an interrupt
             * context.  Only API functions that end in "FromISR" can be used in an
             * interrupt.  Only assert if the critical nesting count is 1 to
             * protect against recursive calls if the assert function also uses a
             * critical section. */
            if( pxCurrentTCB->uxCriticalNesting == 1U )
            {
                portASSERT_IF_IN_ISR();           <--- FAILS!
```

Then

```
extern volatile UBaseType_t uxInterruptNesting;
#define portASSERT_IF_IN_ISR() configASSERT( uxInterruptNesting == 0 )        <--- FAILS!
```

because:
```
/* Records the interrupt nesting depth.  This is initialised to one as it is
decremented to 0 when the first task starts. */
volatile UBaseType_t uxInterruptNesting = 0x01;
```

aggarg, 2025-04-24:
Thank you for explaining. I understand that now. `configASSERT`  and `vAssertCalled`  are defined by the application and it should be okay to update it. Can you try updating `vAssertCalled` to the following:
```
void vAssertCalled( const char * pcFile, unsigned long ulLine )
{
   portDISABLE_INTERRUPTS();
   {
      // your code
   }
   portENABLE_INTERRUPTS();
}
```

If the above definition of `vAssertCalled` came from the Microchip generated code, I do not think it is worth adding this assert as it will break the default generated applications.

KammutierSpule, 2025-04-24:
portDISABLE_INTERRUPTS/portENABLE_INTERRUPTS should work but depends on the implementation (eg: may dump to a USART controlled by some ISR buffer). So it is implementation specific (and I think it is ok).

Since vAssertCalled is on user (Microchip) side, I think should be ok on FreeRTOS project side to add  what is reasonable.

Even with the current vAssertCalled, it will crash me on a stack overflow, which if I backtrack with debug, I will find the `configASSERT( ( ulCompareMatch & 0xFFFF0000 ) == 0 );`, which will help me to fix the bug.

aggarg, 2025-04-24:
> Even with the current vAssertCalled, it will crash me on a stack overflow, which if I backtrack with debug, I will find the configASSERT( ( ulCompareMatch & 0xFFFF0000 ) == 0 );, which will help me to fix the bug.

You are right. I missed that the assert will only fire when configPERIPHERAL_CLOCK_HZ and/or configTICK_RATE_HZ is mis-configured. Would you like to raise a PR?

closed 2025-04-25 by aggarg