FreeRTOS/FreeRTOS-Kernel #1268: [BUG] ARM CR5

foxhoundsk, 2025-04-25:
**Describe the bug**
On target ARM Cortex-R5, I was browsing the ISR path, shown below:

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/03db672b8f45db24aa99f12051f7cf86746b9ed9/portable/GCC/ARM_CR5/portASM.S#L171-L249

Upon entering the ISR, the LR and SPSR are pushed into the IRQ_sp respectively. Afterwards, the processor enters the SVC mode to handle the IRQ accordingly, and right before returning to the interrupted instruction, which typically lays in the System mode, the stack popping operations **seems** incorrectly performed:

```asm
exit_without_switch:
    /* No context switch.  Restore used registers, LR_irq and SPSR before
    returning. */
    POP     {r0-r4, r12}
    CPS     #IRQ_MODE
    POP     {LR}
    MSR     SPSR_cxsf, LR
    POP     {LR}
    MOVS    PC, LR
```

The first `POP {LR}` is correct as the processor still operates in the IRQ mode. However, the second one uses the stack that is mostly not (say it is not nested IRQ) the IRQ_SP (stack pointer of IRQ mode), this can results in the processor jumping to unexpected instruction address due to the unexpected processor mode during the LR popping operation (right after `MSR SPSR_cxsf, LR`).

**Target**
- Instruction Set Architecture: ARMv7-R

**Expected behavior**
I was expecting that, before switching the processor mode back to the interrupted instruction (`MSR     SPSR_cxsf, LR`), the _return address_ previously stored in the IRQ_SP should be popped into the preserved registers such as r5, and use r5 to restore PC after switching back to the interrupted instruction.

**Additional context**

I'm using my knowledge to understand this code snippet, please feel free to criticize me.

Thanks!

aggarg, 2025-04-25:
> However, the second one uses the stack that is mostly not (say it is not nested IRQ) the IRQ_SP (stack pointer of IRQ mode), this can result in the processor jumping to unexpected instruction address due to the unexpected processor mode during the LR popping operation (right after MSR SPSR_cxsf, LR).

Writing to SPSR (Saved Program Status register) does not change the processor mode. SPSR gets copied into CPSR upon returning from interrupt. You can read this for more details - https://developer.arm.com/documentation/den0042/a/Exceptions-and-Interrupts/Exception-handling/Exit-from-an-exception-handler.

foxhoundsk, 2025-04-26:
> Writing to SPSR (Saved Program Status register) does not change the processor mode. SPSR gets copied into CPSR upon returning from interrupt. You can read this for more details - [...]

Thank you for the info! Turns out I was blindly seeing `MSR SPSR_cxsf, LR` as `MSR CPSR_cxsf, LR`, making me to think that the mode has been changed at the time.

closed 2025-04-26 by foxhoundsk