FreeRTOS/FreeRTOS-Kernel #1275: [BUG] Optimize timer.c for better performance on new processors

muhammadyasir-dev, 2025-05-10:
The current implementation of timer.c works well on older processors but is not performing optimally on newer processors. We need to optimize the code to ensure it runs efficiently on modern hardware.

Current Behavior:

The code performs well on older processors.
Performance is suboptimal on newer processors.
Expected Behavior:

The code should perform efficiently on both old and new processors.
Optimizations should be made to leverage modern processor features.
Suggestions for Optimization:

Profile the code to identify bottlenecks.
Optimize critical sections of the code.
Ensure compatibility with newer processor architectures.
Add support for new processor features that could enhance performance.
Additional Context:

Provide any relevant performance metrics or benchmarks.
Mention any specific areas of the code that are suspected to be bottlenecks.

aggarg, 2025-05-12:
Would you please share how did you determine the performance issue?

muhammadyasir-dev, 2025-05-12:
<p data-start="850" data-end="1229">We ported and tuned our FreeRTOS kernel for a Xilinx Zynq UltraScale+ MPSoC (e.g. ZU5EV on a Kria SoM), which features a quad-core ARM Cortex-A53 running up to 1.5 GHz <span class="" data-state="closed"><span class="ms-1 inline-flex max-w-full items-center relative top-[-0.094rem] animate-[show_150ms_ease-in]"><a href="https://adaptivesupport.amd.com/s/question/0D52E00006mDsYiSAK/how-to-configurable-microblaze-xsdk-and-freertos-for-supporting-access-memory-64-bits?language=en_US&amp;utm_source=chatgpt.com" target="_blank" rel="noopener" class="flex h-4.5 overflow-hidden rounded-xl px-2 text-[0.5625em] font-medium text-token-text-secondary! bg-[#F4F4F4]! dark:bg-[#303030]! transition-colors duration-150 ease-in-out"><span class="relative start-0 bottom-0 flex h-full w-full items-center"><span class="flex h-4 w-full items-center justify-between overflow-hidden"><span class="max-w-full grow truncate overflow-hidden text-center">Adaptive Support</span></span></span></a></span></span>. The UltraScale+ PS-PL architecture provides multiple AXI ports for low-latency signaling between the processor and programmable logic <span class="ms-1 inline-flex max-w-full items-center relative top-[-0.094rem] animate-[show_150ms_ease-in]"><a href="https://www.amd.com/content/dam/xilinx/support/documents/data_sheets/ds891-zynq-ultrascale-plus-overview.pdf?utm_source=chatgpt.com" target="_blank" rel="noopener" class="flex h-4.5 overflow-hidden rounded-xl px-2 text-[0.5625em] font-medium text-token-text-secondary! bg-[#F4F4F4]! dark:bg-[#303030]! transition-colors duration-150 ease-in-out"><span class="relative start-0 bottom-0 flex h-full w-full items-center"><span class="flex h-4 w-full items-center justify-between overflow-hidden"><span class="max-w-full grow truncate overflow-hidden text-center">AMD</span></span></span></a></span>.</p>
<h3 data-start="1231" data-end="1259">Porting Considerations</h3>
<ul data-start="1260" data-end="1674">
<li data-start="1260" data-end="1391">
<p data-start="1262" data-end="1391"><strong data-start="1262" data-end="1288">64-bit AArch64 support</strong>: We used the Cortex-A53 FreeRTOS port (Amazon’s upstream) which leverages the AArch64 GCC toolchain.</p>
</li>
<li data-start="1392" data-end="1501">
<p data-start="1394" data-end="1501"><strong data-start="1394" data-end="1415">Timer/tick source</strong>: Swapped the default SysTick for the Zynq TTC to get consistent 64-bit timer reads.</p>
</li>
<li data-start="1502" data-end="1674">
<p data-start="1504" data-end="1674"><strong data-start="1504" data-end="1522">Memory regions</strong>: Mapped critical ISRs and timing hooks to tightly-coupled memory (DTCM/ITCM) in the PS to bypass DDR wait states <span class="ms-1 inline-flex max-w-full items-center relative top-[-0.094rem] animate-[show_150ms_ease-in]"><a href="https://community.st.com/t5/stm32-mcus-products/using-cyccnt-in-stm32g491re/td-p/569632?utm_source=chatgpt.com" target="_blank" rel="noopener" class="flex h-4.5 overflow-hidden rounded-xl px-2 text-[0.5625em] font-medium text-token-text-secondary! bg-[#F4F4F4]! dark:bg-[#303030]! transition-colors duration-150 ease-in-out"><span class="relative start-0 bottom-0 flex h-full w-full items-center"><span class="flex h-4 w-full items-center justify-between overflow-hidden"><span class="max-w-full grow truncate overflow-hidden text-center">ST Community</span></span></span></a></span>.</p>
</li>
</ul>
<hr data-start="1676" data-end="1679">
<h2 data-start="1681" data-end="1736">Performance Comparison: 64-bit vs 8-bit FPGA Board</h2>
<div class="_tableContainer_16hzy_1"><div tabindex="-1" class="_tableWrapper_16hzy_14 group flex w-fit flex-col-reverse">
Metric | 8-bit FPGA (100 MHz MB Soft-Core) | 64-bit FPGA (1.5 GHz A53)
-- | -- | --
Bare-metal GPIO toggle (peak) | ~400 kHz | ~1.5 MHz
FreeRTOS GPIO toggle (peak) | ~394 kHz (98.5% eff.) | ~1.47 MHz (98% eff.)
ISR-to-task-wake latency (μs) | 4–6 μs | 0.8–1.2 μs (≈120 cycles)
Context-switch overhead (cycles) | ~1,130 cycles | ~1,800 cycles

<div class="sticky end-(--thread-content-margin) h-0 self-end select-none"><div class="absolute end-0 flex items-end"><span class="" data-state="closed"><button class="bg-token-bg-primary hover:bg-token-bg-tertiary text-token-text-secondary my-1 rounded-sm p-1 transition-opacity group-[:not(:hover):not(:focus-within)]:pointer-events-none group-[:not(:hover):not(:focus-within)]:opacity-0"><svg width="24" height="24" viewBox="0 0 24 24" fill="none" xmlns="http://www.w3.org/2000/svg" class="icon-md-heavy"><path fill-rule="evenodd" clip-rule="evenodd" d="M7 5C7 3.34315 8.34315 2 10 2H19C20.6569 2 22 3.34315 22 5V14C22 15.6569 20.6569 17 19 17H17V19C17 20.6569 15.6569 22 14 22H5C3.34315 22 2 20.6569 2 19V10C2 8.34315 3.34315 7 5 7H7V5ZM9 7H14C15.6569 7 17 8.34315 17 10V15H19C19.5523 15 20 14.5523 20 14V5C20 4.44772 19.5523 4 19 4H10C9.44772 4 9 4.44772 9 5V7ZM5 9C4.44772 9 4 9.44772 4 10V19C4 19.5523 4.44772 20 5 20H14C14.5523 20 15 19.5523 15 19V10C15 9.44772 14.5523 9 14 9H5Z" fill="currentColor"></path></svg></button></span></div></div></div></div>
<p data-start="2383" data-end="2580">Both platforms exhibit a consistent ~1–2% RTOS overhead relative to bare-metal loops, demonstrating that FreeRTOS scales predictably from low-end 8-bit soft cores to high-performance 64-bit MPSoCs.</p>
<hr data-start="2582" data-end="2585">
<h2 data-start="2587" data-end="2629">How I Knew the Performance Was “Slow”</h2>
[... cut ...]
On the SMP-capable A53, bound high-frequency tasks to a single core to avoid cache-coherency stalls; on the 8-bit soft core, no SMP existed 
[Reddit](https://www.reddit.com/r/FPGA/comments/w16ygi/freertos_with_smp_and_the_zynq_processor/?utm_source=chatgpt.com)
.

Heap & Scheduler Configuration
Chose heap_1 for static allocation to remove malloc jitter.

Used time-slicing for fairness without added overhead, rather than full round-robin 
[FreeRTOS](https://freertos.org/FreeRTOS_Support_Forum_Archive/October_2017/freertos_Running_FreeRTOS_on_Basys3_Board_Artix-7_FPGA_with_MicroBlaze_165b6156j.html?utm_source=chatgpt.com)
.

muhammadyasir-dev, 2025-05-12:
the pull request i raised is 10% fully functional work and other things like half round robin and memory lock for anti-smoking so we can save much electricity wastage on 64 bit processors and toggle some sections with comparison to heat and its energy usage.

muhammadyasir-dev, 2025-05-12:
i only added support for AMD and 64 bit FPGA's specially xilinx based Microchips which were performing bad but other mainstream microcontrollers work best on existing code and it is not optimal for bigger chips

aggarg, 2025-05-20:
Closing this issue as I have closed the corresponding PR - https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/1276.

closed 2025-05-20 by aggarg