FreeRTOS/FreeRTOS-Kernel #1277: [BUG] Compilation fails with IAR/RL78 porting

5iBri, 2025-05-12:
**Describe the bug**
Compilation fails when compiling the file portasm.s. The following errors are generated : 

[image]

**Target**
- IDE : IAR Embedded Workbench for Renesas RL78 Version 5.10.3
- IAR/RL78 port

**To Reproduce**
- Use the project demo.ewp
- compile it. It generates errors.

**Expected behavior**
Compilation must be succeed

5iBri, 2025-05-12:
Here is the IAR project : [demo.zip](https://github.com/user-attachments/files/20160678/demo.zip)

kar-rahul-aws, 2025-05-12:
Hi @5iBri 
Thank you for pointing out the issue. The solution you suggested looks good, once reviewed and merged, you can close this issue.

closed 2025-05-12 by aggarg