FreeRTOS/FreeRTOS-Kernel #1296: [BUG] configENABLE_MPU = 1 -> Redefinition of xEventGroupClearBitsFromISR / xEventGroupSetBitsFromISR

mu-hp, 2025-07-22:
**Describe the bug**
When I enable the  mpu, xEventGroupClearBitsFromISR and xEventGroupSetBitsFromISR does not use the MPU wrapper.

```
/home/<snip>/build/debug/<snip>/gcc/_deps/freertos-src/include/./event_groups.h:459:13: warning: "xEventGroupClearBitsFromISR" redefined
  459 |     #define xEventGroupClearBitsFromISR( xEventGroup, uxBitsToClear ) \
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/<snip>/build/debug/<snip>/gcc/_deps/freertos-src/include/./portable.h:105,
                 from /home/<snip>/build/debug/<snip>/gcc/_deps/freertos-src/include/./FreeRTOS.h:118,
                 from /home/hp/repos/vps/sw-basic-next/prj/src/app/app.cpp:11:
/home/<snip>/build/debug/<snip>/gcc/_deps/freertos-src/include/./mpu_wrappers.h:215:21: note: this is the location of the previous definition
  215 |             #define xEventGroupClearBitsFromISR    MPU_xEventGroupClearBitsFromISR
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/<snip>/build/debug/<snip>/gcc/_deps/freertos-src/include/./event_groups.h:615:13: warning: "xEventGroupSetBitsFromISR" redefined
  615 |     #define xEventGroupSetBitsFromISR( xEventGroup, uxBitsToSet, pxHigherPriorityTaskWoken ) \
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/<snip>/build/debug/<snip>/gcc/_deps/freertos-src/include/./mpu_wrappers.h:216:21: note: this is the location of the previous definition
  216 |             #define xEventGroupSetBitsFromISR      MPU_xEventGroupSetBitsFromISR
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
```

**Target**
- stm32h5x (cortex-m33, armv8)

Observed with FreeRTOS versions:
- V11.2.0 GITHUB_REPOSITORY "FreeRTOS/FreeRTOS-Kernel"
- GIT_TAG 684b34ca8db0de1277b9b5359e0d1d822b53b6a2 GITHUB_REPOSITORY "FreeRTOS/FreeRTOS-Kernel"

**Host**
- Debian 12

**To Reproduce**
- enable MPU
```
#define configENABLE_MPU 1
```

- include <event_groups.h>
```
#include <event_groups.h>
```

**Expected behavior**
- no warnings (afaik MPU wrapper should be used -> MPU_xEventGroupClearBitsFromISR() / MPU_xEventGroupSetBitsFromISR()

kstribrnAmzn, 2025-07-24:
Have you defined `configUSE_MPU_WRAPPERS_V1` to `1`? This warning should not occur if you're using MPU wrappers V2 (which is the default).

mu-hp, 2025-07-25:
no, I have:

```
#define configUSE_MPU_WRAPPERS_V1 0
```

**1st definition of ```xTimerPendFunctionCallFromISR``` in mpu_wrappers.h** (line 215)
```
        #if ( configUSE_MPU_WRAPPERS_V1 == 0 )
            #define xEventGroupGetStaticBuffer     MPU_xEventGroupGetStaticBuffer
            #define xEventGroupClearBitsFromISR    MPU_xEventGroupClearBitsFromISR
            #define xEventGroupSetBitsFromISR      MPU_xEventGroupSetBitsFromISR
            #define xEventGroupGetBitsFromISR      MPU_xEventGroupGetBitsFromISR
        #endif /* #if ( configUSE_MPU_WRAPPERS_V1 == 0 ) */
```

**2nd (re-)definition of  ```xTimerPendFunctionCallFromISR``` in event_groups.h** (line 459)
-> does not depend on ```configUSE_MPU_WRAPPERS_V1```, ```configUSE_TRACE_FACILITY``` is 0 (default) 
```
#if ( configUSE_TRACE_FACILITY == 1 )
    BaseType_t xEventGroupClearBitsFromISR( EventGroupHandle_t xEventGroup,
                                            const EventBits_t uxBitsToClear ) PRIVILEGED_FUNCTION;
#else
    #define xEventGroupClearBitsFromISR( xEventGroup, uxBitsToClear ) \
    xTimerPendFunctionCallFromISR( vEventGroupClearBitsCallback, ( void * ) ( xEventGroup ), ( uint32_t ) ( uxBitsToClear ), NULL )
#endif
```

similar case for ```xEventGroupSetBitsFromISR```

AhmedIsmail02, 2025-08-14:
@mu-hp Could you please confirm if the changes on [event-groups: Fix multiple definitions #1307](https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/1307) fix the issue raised?

mu-hp, 2025-08-14:
@AhmedIsmail02 same repo and branch but 8cdf4e9b1f96b2a71a3d5120276213b14507c8d0 is still working, no change for me.

closed 2025-08-14 by kstribrnAmzn