FreeRTOS/FreeRTOS-Kernel #1298: [Feature Request] PIC32 port, implement xPortIsInsideInterrupt

KammutierSpule, 2025-07-25:
There is no xPortIsInsideInterrupt implemented for PIC32 ports.

There is a uxInterruptNesting but it is not clear its usage:
`extern volatile UBaseType_t uxInterruptNesting;`

kstribrnAmzn, 2025-07-25:
How does https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/1300 work for you?

KammutierSpule, 2025-08-07:
> How does [#1300](https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/1300) work for you?

I didn't tried yet, I may test on next week.
I left comments on your PR.

kstribrnAmzn, 2025-08-11:
Thanks for further testing this out and letting me know about the issues. I got MPLAB setup myself and polished the hasty code I wrote. It should now build without warnings. Let me know if it's working for you

KammutierSpule, 2025-11-27:
> I have updated my PR to check the EXL bit of the status register. Feel free to try it out.

It didn't worked for me.

I tested again and this is what I experience:
```
$12
ulCurrentInterrupt
outside ISR: 0x25000001

inside ISR: 0x25000C01
or
inside ISR: 0x25000401
```