FreeRTOS/FreeRTOS-Kernel #1305: [Feature Request] <Support for Cortex-M52>

JidongMei, 2025-08-11:
**Is your feature request related to a problem? Please describe.**
So far, FreeRTOS portable dir add Cortex-M55 and Cortex-M85 support .But Cortex-M52 is not on the list 

**Describe the solution you'd like**
Do you have plan for adding Cortex-M52 support? 
I'm form ARM China,Cortex-M52 CPU software teams. If Cortex-M52 support plan is not on the way . Maybe I can  contribute to do this work.

JidongMei, 2025-08-12:
> Are there any architecture differences that would necessitate the need for a separate port? Or would the port be same as M55 only copied to a separate M52 directory?

No architecture differences. The port be same as M55 and only need copied to a separate M52 directory.

AhmedIsmail02, 2025-08-12:
Hello @JidongMei,

Thank you for submitting this enhancement feature, I think the only architectural difference between Cortex-M55 and Cortex-M52 when it comes to FreeRTOS is that Cortex-M52 has optional PACBTI extension support. I would think of the changes to be as follows:

* Creating new directories for Cortex-M52 with files copied from Cortex-M55 port.
* Adding the necessary changes to the copied files (e.g., changing the `portARCH_NAME` in [portmacro.h](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/GCC/ARM_CM55/non_secure/portmacro.h) to `Cortex-M52`).
* Adding the port to the [top-level CMakeLists.txt](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/CMakeLists.txt).
* Adding the port to the [portable CMakeLists.txt](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/CMakeLists.txt).
* Adding the port to the [copy_files.py](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/ARMv8M/copy_files.py) python script.
* The `README.md` under https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/ARMv8M/ to be updated to include Cortex-M52 port.

closed 2025-12-19 by aggarg